paulscottrobson/neo6502-firmware #43: Latest  USB release (2381) not working

leventp, 2024-01-25:
Installed the latest USB release 2381. The boot up screen shows fine but no response from the keyboard accepted.
Back to release 2197. All seem to work. So looks like there is an issue with the latest USB release.
Cheers,
Levent

leventp, 2024-01-26:
It displayed all the way to the prompt (including build # and Wozmon char). 
By the way, the link you included does not contain the USB  firmware serial check disabled.

n2kbg, 2024-01-26:
FYI- the latest SD card version doesn't work for me. I don't have a working USB hub and didn't try the USB build.

With the SD version, I do get the splash screen and the wozmon prompt but no keyboard input. The previous SD version works correctly with the same keyboard/hardware.

I'm using a rev A board if that matters.

paulscottrobson, 2024-01-26:
It shouldn't do. It's odd as I haven't touched the USB code in ages, or the SDCard code. I will put up a release that has no serial code in it (the main change) and see if that's any better.

paulscottrobson, 2024-01-26:
Okay release 11 is being built now. This has two changes ; one disables the serial code and the other is to reduce the allocated memory for graphics objects. Previously this has caused boards to crash and the compiler gives no warning.

Unfortunately if I can't replicate the problem I can only fix it by removing code until it works. Which is a pain, sorry.

leventp, 2024-01-26:
USB Release 11 (version 2448)  works for me.

paulscottrobson, 2024-01-26:
Okay if release 11 worked can you try release 12. This has the serial code put back in, but accessed differently. You have to type / [ENTER] at the command line to start it manually.  It also does _nothing_ with the serial code until it's actually needed. 

I'm pretty sure the serial code is the problem. Why, I don't know. I suspect it's some timing or interplay between the USB system and the UART system.

If it all seems to work, can you try / and let me know what, if anything happens.

leventp, 2024-01-26:
Release 12 booted up  into BASIC OK. Entering  '/' displays 'Serial link enabled.'. After that the system hangs. No response from the keyboard.

n2kbg, 2024-01-26:
Found a USB/TTL serial cable for more testing.

Release 12 (2470) for SD booted into basic. As with leventp when I enter '/' it displays the serial link message and then does not accept any more input from the keyboard. It does not display anything on the serial port either.

However, I found that on release 10 (2381) the USB keyboard does work *IF* a the serial port is active! If the serial port is not connected then they keyboard does not work.

If I boot 2381 with the serial port connected and then remove the serial connection then the usb keyboard will stop working as soon as the serial is removed.

paulscottrobson, 2024-01-27:
That's what I expected. I had to do it in a bit of a rush, so I didn't add the ESC out of it. It needs some work on timing out.  It works well enough, the problems occur when it gets out of sync. Rather than timing out and exiting it is waiting for packet completion, I should let it time out once a packet has started and reject the incomplete packet and error.

leventp, 2024-01-27:
So far, the Buck Rogers release working fine. 
Just a question. I see 'CASE' and 'ENDCASE' tokens  defined. Are they implemented?
Cheers,
Levent

paulscottrobson, 2024-01-28:
I've closed this as I've had no further reports and the broken ones seem to work. It may still have consequential issues, the serial port might not work maybe, but if that's an issue it can be reopened.

closed 2024-01-28 by paulscottrobson