paulscottrobson/neo6502-firmware #44: Enhancement: Provide additional commands for  file operations

leventp, 2024-01-25:
Hi Paul,
Thanks for the excellent Neo BASIC implementation. I just started playing with it. It looks like we can issue a CAT command to see root directory contents of the attached device (using USB version). It would be useful to have CD and MKDIR commands as well. 
Keep up the good work.
P.S. Not sure how to browse sub directories at the moment.

paulscottrobson, 2024-01-26:
You can't :) It's something that will be added

leventp, 2024-01-26:
As you indicated, it would be nice to implement a MOS like interface similar to the one developed for Agon Light 2.

paulscottrobson, 2024-02-09:
This will be delayed until @davidgiven 's file system routines for CPM/65 are stable. It may even be done by CPM/65 itself as an option.

PR #122 (2024-02-15): Add a more complete filesystem API.
As discussed on Discord.

This adds a mostly-complete set of file APIs for general purpose file reading and writing through handles, directory operations, CWD, etc. It's had some testing and appears to mostly work.

There are omissions:

- there's no way of setting file attributes
- all errors are reported as 1 --- we really need to define some and do a mapping
- documentation updates

I'm sure there are others.

Fixes: #44
closed 2024-02-15 by paulscottrobson