paulscottrobson/neo6502-firmware #45: Add capability to set up different keyboard options

leventp, 2024-01-26:
Looks like the default keyboard is that of UK. It would be nice to set up different keyboard options (at least US ;)

paulscottrobson, 2024-01-26:
This is actually already there. I can use UK keyboards and adding new ones is fairly straightforward. However, SDCard will not load the configuration file first up, I have no idea why as yet.

leventp, 2024-01-26:
What is the name of the configuration file. I am using USB stick. I can change it to the US version and see what happens.

paulscottrobson, 2024-01-27:
You can do it now. There is a program "config.bas" which writes a file called ".config" to the storage. It works fine if this file is absent. It should be the US version by default, if there are any mistakes let me know, as I have a GB keyboard, I'm British.

Keymaps are hard coded into the firmware at present. The plan was to use US as a base as most keyboards aren't far off except for places like Greece, Malta, China, some of Eastern Europe I think. About the worst in France which still uses AZERTY. There's a problem with reading the .config file.

paulscottrobson, 2024-01-29:
This is being redeveloped today. While I am away next week Sascha Schneider (damn_pastor) will create a german version to check everything works. Once that gets back people will be able to create their own keyboard locale descriptors which will go into the firmware.

paulscottrobson, 2024-01-29:
This is now present ; example locale files are here https://github.com/paulscottrobson/neo6502-firmware/tree/main/firmware/common/locale - please note at present we are limited to the standard ASCII characters from 32-127

closed 2024-01-29 by paulscottrobson