paulscottrobson/neo6502-firmware #48: CLS side effect with turtle graphics

leventp, 2024-01-28:
If CLS is issued after a few turtle graphics commands and subsequently  when a new turtle command is issued (see the code below), the previous and current turtles are both displayed on the screen. TURTLE CLEAR (after CLS) does not rectify this situation either (as it does not clear the screen).

CLS
TURTLE CLEAR
FORWARD 50
CLS 
LEFT 90 : REM 2 turtles

I think TURTLE CLEAR should  also clear the screen and reset its position.. Perhaps a new command like TURTLE ORIGIN can return the turtle to its starting position without clearing the screen.

paulscottrobson, 2024-01-29:
There's a bug here, in that CLS should behave differently when there are sprites present (it should only clear the text/graphics area, not the sprites as well, which is done by SPRITE CLEAR). It does the whole screen, so when the system attempts to remove the sprite it redraws it (because of Xor sprites).

The bitmap display splits into 2 layers using the palette when sprites are on.

I think you're right about the commands, that's much better. I was never very happy with it.

paulscottrobson, 2024-01-30:
This is now fixed. CLS clears the graphics layer. The Sprite Layer is reset and cleared either by SPRITE CLEAR or by CLEAR/RUN. Previously this just reset the data, which would cause sprite records regarding drawn state to be inconsistent. It also means TURTLE should behave properly.

closed 2024-01-30 by paulscottrobson