paulscottrobson/neo6502-firmware #52: Release considerations

leventp, 2024-01-30:
Some consideration when making releases:

- It is quite a common/standard  practice to link any release content to the corresponding issue for tracking purposes. 
- It is also common practice to create pull requests before making releases so that the code to be integrated can be reviewed 

I know these will add additional time and effort but in the long term the benefits will be valuable.

Just my 2 cent (pennies ;)

Levent

paulscottrobson, 2024-02-03:
Do you know a good explainer for this process ? I basically used GitHub as a backup system so I'm catchup here !

leventp, 2024-02-03:
I think you follow Agon Light MOS/Basic/VDP development on Github. They use the process I mentioned. See the link below.

https://github.com/breakintoprogram/agon-vdp/releases

astralaster, 2024-02-03:
The main difference is to work on branches other than main. Then you can create a PR to merge it to main and the rest will work automatically. I also support this idea. It gives greater transparency.

paulscottrobson, 2024-02-04:
Sounds good to me. I'm getting there .... Slowly

paulscottrobson, 2024-02-06:
I will start this shortly.

paulscottrobson, 2024-02-07:
One thing. I notice with Agon they list the PRs and their authors. It looks like it's done automatically but I couldn't find out how it was done.

leventp, 2024-02-07:
When you make a release, 'Generate Release notes' unlocked once you create a new label.

closed 2024-02-07 by paulscottrobson