paulscottrobson/neo6502-firmware #53: Add  function definition

leventp, 2024-01-30:
It would be useful to have function definitions similar to that of PROC.  For example:

func add(a,b)
  add = a+b
endfunc

print add(3,5)

paulscottrobson, 2024-01-31:
This is slightly challenging. The interpreter has a stack, to deal with operate precedence and also to store parameters both to procedures and function calls. This is quite efficient as it uses the 'X' register as the stack index. 

The downside is that a function call needs to maintain this stack state. The likely solution to this is that used in the Atari language Action! which does almost the exact same thing for efficiency, in fact more so as it knows what goes where on the stack as its compiled, but deals with functions by pushing the entire part completed calculation on the BASIC stack which is used for things like for/next, procedures, locals and is slower and larger. This is inefficient.

A possibility as a limited compromise is to only allow it at the base level, so there is no intermediate calculation in progress. So you could do print add(3,5) and print add(3,5)+10 but not print 10+add(3,5) because in that calculation the 10 has been pushed on the stack already. 

It is one of those choices I had to make, I knew when using an X-stack that it would handicap user defined functions. 

I'll leave it as an open enhancement for now.

leventp, 2024-01-31:
Another option would be passing parameters by reference to procedures.

paulscottrobson, 2024-01-31:
Actually that's a much better idea :)

leventp, 2024-02-04:
Looks like all variables are global and can be set inside a procedure.  It is a workaround ;)

paulscottrobson, 2024-02-05:
Yes, local pushes them on a stack

closed 2024-02-08 by paulscottrobson