paulscottrobson/neo6502-firmware #54: Allow png images to be loaded as sprites

leventp, 2024-01-30:
At the moment, creating sprite images is a rather unusual process (need to run 2 python scripts to start with). I was wondering whether it is possible to directly load 'png' images as sprites.

paulscottrobson, 2024-01-31:
It's possible, certainly. The problem with it is you don't have one PNG you have 20 of them, and that's slower than loading in the single file. It could be done, yes. 

There's no standard really for sprite and tile sheets that I'm aware of. Every system tends to have its own. Sometimes every developer has their own. There is tiled for tile maps and this has some sprite sheet support built in. Usually what happens is there is a rectangular PNG chopped up into 32x32 or whatever cells, basically the same as what we have now except there's no gap between the boxes. There are some commercial products like TexturePacker but I didn't really want people to have to buy things. Even then you'd have to write an exporter/converter.

There's only really one Python script, that combines the 3 files into a gfx file, the other creates the blanks. The reason for choosing Python is partly its cross platform portability, but primarily the Python Imaging Library ; having a library which reliably can process images cross platform and format is a godsend. It also provides a library for importing graphics so it can be done automatically. 

Having a system whereby there are a collection of either individual tile/sprite pngs or sheets of them and these are combined rather than the 'boxes' file is fairly straightforward.

paulscottrobson, 2024-02-09:
All suggestions welcome.

bocianu, 2024-02-12:
I'm just curious, when you will add that header to provide automatic load, will it be possible to use this feature, to load sprites. Similar to load, when you will set destination adress as $ffff it would load data block to sprite memory.
It could be nice feature connected with this multiblock feature. In a single file you can distribute all gfx, sprites, and your binary executable as well. For now you have to deliver sprites to your application in a separate file, and it is not very convenient.

paulscottrobson, 2024-02-15:
Maybe tomorrow. I liked the look of the .o65 format, albeit in a limited format.

closed 2024-02-23 by paulscottrobson