paulscottrobson/neo6502-firmware #55: Some Control Codes are not working

leventp, 2024-01-30:
The  following Control Codes (described in the api document) are not working for me.

CTRL+E  : No insert effect
CTRL+G : No effect 
CTRL+F : Page down? Not sure what this does
CTRL+X : Draws a pipe char as cursor first time then no effect
CTRL+Z : No effect
CTRL+[ : prints "{"

paulscottrobson, 2024-01-31:
It's a documentation issue.  Console codes have two functions ; they are keyboard maps - so INSERT should generate the Ctrl+E code, and they are console out functions. 

Insert needs implementing. At present you can't insert a character in a line when editing on the screen and this is an issue. Removing matters much less because spaces are removed from the tokenised code anyway. 

Hence the change of the title.

leventp, 2024-02-02:
Note that Backspace is not working either. Only blanking out the character next to the left of the cursor. Making editing lines of BASIC  a bit difficult.

paulscottrobson, 2024-02-03:
Same issue. Apart from strings doesn’t matter because spaces aren’t stored anywaySent from my iPhone

paulscottrobson, 2024-02-07:
Insert and Delete and Backspace now work.

closed 2024-02-07 by paulscottrobson