paulscottrobson/neo6502-firmware #62: All variables has a global zero (0) index pointer

leventp, 2024-02-04:
It is possible to assign values to any variable at zero index (without  DIM). In this case,  all point to the same memory location (thus contain the same assigned value). Not sure if this is a feature or bug;). CLEAR and NEW will not affect the current value.
Code examples:
a=5
a(0) = 10  ' any zero index will now contain 10
print a, a(0)
5    10
print hello(0)
10
print foo(0)
10
changing(0)=42   ' any zero index will now contain 42
print a(0)
42

CLEAR   
print a(0)
42

paulscottrobson, 2024-02-05:
Definitely a bug !

paulscottrobson, 2024-02-07:
This is what I expected and is definitely in the 'dumb' category. I forgot to check the array has actually been dimensioned, so the array thinks it is at memory location zero with a 0 element array. This is why hello(0) foo(0) all return the same number. 

Now checks for this address being zero and produces an appropriate error.

closed 2024-02-07 by paulscottrobson