paulscottrobson/neo6502-firmware #63: custom turtle sprite is not working

bocianu, 2024-02-05:
Maybe I'm doing something wrong, but I'm unable to replace turtle sprite using API call  9,1.
My sprite is 16x16, index 0. GFX Resource is properly loaded because I can use DrawImage to put in on the screen.
But when I select it (on P0) in turtle Init function, the default sprite is not replaced. 
I tried indexing it as 0, 128, 192 - nothing works for me.

paulscottrobson, 2024-02-06:
Currently there is no option to have a custom turtle. It's not a bad idea though so I will change this appropriately.

bocianu, 2024-02-06:
Api documentation says it is custom sprite option :) that's why I had asked.
[image]

paulscottrobson, 2024-02-07:
Unclear. What it means is that's the sprite to use for the turtle. It co-opts one for its own use. I'll improve it.

closed 2024-02-07 by paulscottrobson