paulscottrobson/neo6502-firmware #65: Synchronization of drawing with the framerate of the display device

bocianu, 2024-02-05:
In writing games and demos, it is quite important to be able to synchronize to the device framerate in order to maintain the right smoothness of movement and animations, with no hiccup,  glitches while loosing sync.
How does this currently look in the Neo6502? A virtual device emulated by RP2040 is responsible for the display, so
as what frequency does it generate the image? Would it be possible to somehow synchronize with its cycles?
This would be a very useful feature from the point of view of smoothness of animation generation.

For now, it is possible to watch the clock given by the API and draw the picture when we have the last bit is set to zero.
This gives us a frequency of 50Hz, but I don't know at what actual frequency is the image generated by RP2040?

The options I can think of are:
- an interrupt triggered when the frame is finished drawing (like vertical blank interrupt in actual retro hardware), 
- API command returning a result only after the current frame is finished drawing. 

The latter solution is a bit worse, because it forces the current program to stop working. 
But i'm not an hardware expert so I do not even know if it is possible at all (from rp2040 point of view).
Maybe there are some other options?

paulscottrobson, 2024-02-07:
I don't think it's practically possible to fire interrupts. The hardware is sort of there but it will break UEXT and probably SDCard as well.  This leaves the option of something like wait for vertical blank, which should be feasible. 

Maybe just VBLANK ?

paulscottrobson, 2024-02-09:
I suppose you could have a HASVBLANK() , this would return TRUE if a VBlank had occurred between this call and the previous call, if there was one ?
What do you think ?

bocianu, 2024-02-09:
that should be enough I think. Or maybe a vblank counter working tha same way? when it changes it's value you knot that it just occured between reads. And counter let's you identify I you skipped frame or not.

paulscottrobson, 2024-02-09:
VBLANKS() returns this, it is updated immediately after the rendering of the final line.

closed 2024-02-09 by paulscottrobson