paulscottrobson/neo6502-firmware #66: areas occupied and used by basic.

bocianu, 2024-02-05:
If I want to write programs (not in basic), but run from the basic interpreter - I would like to know where I can safely locate to not overwrite some sensitive areas. This will also allow me to safely exit the application when it is finished.

I'm missing information (maybe it's in the documentation, but I haven't found it) about what areas of memory are occupied and/or actively used by basic, and all the software loaded on startup.  The only think I found is that the basic binary starts at $800.

paulscottrobson, 2024-02-07:
So, the memory copy from flash (just an memcpy) won't work if the display has been enabled. 

dispatch.cpp starts

```
	MEMInitialiseMemory();                                         // Set up memory, load kernel ROM
	GFXSetMode(0);                                                              // Initialise graphics
	SPRReset();                                                                 // Reset sprites.
	LOGDrawLogo();                                                              // Draw logo
```
and won't work if you cross the first two over, even though they are not connected. It appears to be able to download some data from the flash, but if you draw too much it crashes. This suggests some sort of problem with the picodvi timing which I haven't yet fathomed and may need specialist help on.

So the answer is you should reinitialise basic through API call group 1 function 3, then do a jmp (0) , but it doesn't work, and I can't fathom out why.

As regards the general issue, the intepreter does start at $800, variable and BASIC starts at PAGE and stack and string memory starts at $F000. If you want to co-exist with BASIC, the CLEAR command allows a parameter which is the highest location BASIC will use, so CLEAR $D000 means the stack and strings will grow down from $CFFF. It is not possible to set this as a default.

Practically, if you push it in upper memory then BASIC will not be harmed. If you exit via jmp $800 this will cold start BASIC which will reinitialise the stacks and clear variables. For it to survive you just have to avoid memory below PAGE. Page is currently at $3900, this is going up slowly as features are added to BASIC - not that quickly because many of the features are in the API not the interpreter. 

I shall leave this open as the 'reboot BASIC' code is still broken.

bocianu, 2024-02-07:
Thank You for explanation. I think that information should be also included in basic.pdf document for future use.

n2kbg, 2024-02-09:
This popped up in a blog I read recently so I thought I'd pass it on. RP2040 memory allocations and execution core have interactions. This may be irrelevant to this issue but even so is probably good to know in the back of your mind since this neo6502 is really pushing the RP2040's capabilities.

https://petewarden.com/2024/01/16/understanding-the-raspberry-pi-picos-memory-layout/

closed 2024-02-28 by paulscottrobson