paulscottrobson/neo6502-firmware #72: INSERT enhancement

leventp, 2024-02-08:
At the moment, INSERT (CTRL+E) enables insertion of one character at a time. When editing a BASIC program line, this is a bit cumbersome. If possible, it would be desirable to keep INSERT mode on until another control code is entered.

paulscottrobson, 2024-02-08:
Slightly problematic because of the way the insert works, but not impossible. Though I'd rather accept this as is, and work on the screen editor design which would behave normally.

leventp, 2024-02-08:
Great. Looking forward to this.

paulscottrobson, 2024-02-09:
I think it will be a generic editor for everything, so it would have an API to request line text, insert, delete and so on. Not quite sure yet.

leventp, 2024-02-09:
One thing I like about the current screen editor is that it is very flexible (Commodore 64 like) and I can go anywhere on the screen make changes and apply them instantly. Can issue a cat command and load any file on the screen (with multiple inserts to make room for the load command). I suggest the new editor should be invoked by a command (say 'edit') so we won't lose the existing editing capabilities.