paulscottrobson/neo6502-firmware #91: Have some sort of blitter functionality between memory elements

paulscottrobson, 2024-02-10:
All ideas/models working. Bear in mind whilst it could use DMA, it does have to run on the emulator as well.

paulscottrobson, 2024-02-10:
@astralaster: I think DMA should be possible to add to the interface. The reverse direction could also be interesting. I would just add DMA to the interface with a hidden offset to the framebuffer. Maybe start with just using the CPU to copy.Like offset in memory, byte count and offset in framebuffer? The question about DMA like feature is, should there be more accessible than the framebuffer?
It could also be completely generic if you give out the offset to the framebuffer. But also kind of dangerous.
@paulscottrobson Well, there are three serious chunks of memory ; the 64k 65C02 space, the 75k display space, and the graphics memory space (currently 32k). I'd be inclined to lock it in there as it's a little risky. I'm also slightly wary because of the weird bug where you can't copy the BASIC image from Flash to RAM once the screen is initialised.

zbyti, 2024-02-10:
https://github.com/tebe6502/Mad-Pascal/tree/master/samples/neo6502/slideshow this example will be much more faster ;)