paulscottrobson/neo6502-firmware #92: Add a header to a file to provide an automatic load and optional execute functionality

paulscottrobson, 2024-02-10:
Extend the file loader to support it, extend RUN so it can run BASIC and Code files directly

bocianu, 2024-02-11:
I think this approach is quite good, and works since 70's ;)

https://www.atarimax.com/jindroush.atari.org/afmtexe.html

paulscottrobson, 2024-02-12:
Would like an optional autorun ?

bocianu, 2024-02-12:
what do you mean by autorun?  autostarting some programs from SD or USB i they are named  for example "autorun.neo" ? 
Or just starting loaded binaries?
For both options I vote yes :D

paulscottrobson, 2024-02-12:
$FFFF header
xxxx load address
xxxx size of data to load
xxxx execute address ($FFFF => do not execute)

Would be interested in @davidgiven 's views on this as it relates to CPM/65 formats ?

davidgiven, 2024-02-12:
It doesn't affect CP/M-65 binaries as they have their own header --- the only file which would be affected is the bootstrap loader itself, which is a native neo6502 binary.

Regarding the format itself: you might want to consider using an existing binary format; Fuzix has a simple flat header: https://github.com/EtchedPixels/FUZIX/blob/master/Kernel/include/exec.h There's also O65, which is more complicated and can optionally support dynamic linking. http://www.6502.org/users/andre/o65/index.html The advantage is that tooling already exists for these.

(I'd also suggest using a clearly distinct magic number, to avoid false positives when detecting binaries! I made that mistake with CP/M-65, which doesn't have a magic number...)

paulscottrobson, 2024-02-12:
Thanks, I will have a nose about. I suspect we won't get as far as dynamic linking but I'll have a peek.

bocianu, 2024-02-12:
> $FFFF header xxxx load address xxxx size of data to load xxxx execute address ($FFFF => do not execute)
> 
> Would be interested in @davidgiven 's views on this as it relates to CPM/65 formats ?

Will it be possible to combine multiple blocks in one executable?
for example to load one small block on page0, another data block above $A000 and main code block on $800 and run it?

davidgiven, 2024-02-12:
IIRC, O65 can do that (it's extremely flexible; note that you don't have to implement _all_ of it). Look for 'chain bit' in the docs.

davidgiven, 2024-02-14:
Actually, I completely forgot to say... there's always ELF! llvm-mos has a set of extensions to allow object files for the 6502, but if you just want to load programs you don't need any of that --- ELF executables are deliberately designed to be easy to load; you just follow some pointers until you locate the program header table, and that will give you a list of segments plus file offsets plus size and address, and you just read them in. It has the advantage of being thoroughly well understood and tooling already exists for it.

paulscottrobson, 2024-02-15:
O65 looks good. It could be partially implemented easily enough, and could be extended later to include all the fancy stuff, though it's really a bit OTT for the 6502 :) It looks like in its rawest form it could just be a binary blob with a larger than normal header (which could be chained to allow multiple files).

paulscottrobson, 2024-02-15:
o65 doesn't seem to have an autorun option.

davidgiven, 2024-02-15:
There's a bit in the header to mark the file as being executable. However, I don't spot an entry point address, which is weird. Maybe it's assuming that the OS will find it via the globals table, which is a bit unfriendly.

paulscottrobson, 2024-02-16:
Suggestion for header, having looked at the suggestions and alternatives.
```
#
#		Magic Numbers. 03 is an invalid 65C02 opcode the others are NEO. A Basic program cannot start with $03 either.
#
03,3E,35,4F
#
#		Control Byte. Bit 7 is continuation ; if set, there is another block after this one. All others are zero by default.
#		and are for extensions.
#
cc
#
#		Load Address (16 bit). $FFFF means load to graphic object memory. $FFF9,B and D are reserved for extension.
#
llhh
#
#		Load Size (16 bit). Size of data that follows. Can be zero.
#
llhh
#
#		Execute address. $FFFF if block not executable
#
llhh
#
#		Data block (if load size non zero)
#
[data]
#
#		If the continuation bit was set, a second control byte etc. follows immediately.
#

#
#		It is not possible to affect the execution behaviour of the 65C02. Therefore the Execute code puts code in the parameter block
# 		either JMP llhh (if execute address is not $FFFF) or RTS (if it is)
#
#		Thus on exit from the load routine, assuming no error, JSR $FF04 (the parameter block) will either run the code as required or
# 		alternatively have no effect.
#
```

davidgiven, 2024-02-16:
Looks good to me --- my only quibble is that I'd suggest adding one, maybe two more bytes just after the magic number to for firmware identification, currently defined to be zero. I'm sure eventually we're going to have multiple firmwares with different feature sets (e.g. my very hypothetical 80-column one) and having a way for the binary to declare that it requires a particular firmware will likely be useful.

paulscottrobson, 2024-02-18:
This is now working - I think :) and will be release shortly as v14.0.0 - there's a helper app using the same syntax as the emulator (e.g. python exec.zip frogger.gfx@ffff frogger.bas@page exec).

closed 2024-02-18 by paulscottrobson

bocianu, 2024-02-18:
Maybe emulator should also be able to recognize header in file loaded from commandline? now I have to keep 2 versions, one without the header if I want to run it directly from windows, and header version to be loaded from emu. Thanks in advance!