paulscottrobson/neo6502-firmware #94: problem building according to documentation

bill-auger, 2024-02-10:
i had trouble compiling 'neo6502-firmware'
```
$ gcc --version
gcc (GCC) 13.2.1 20230801

$ 64tass --version
64tass Turbo Assembler Macro V1.59.3120
```

firstly, according to the wiki "Building-Firmware", the build script given 'build_all.sh' does not exist; so i could not do that step

next, according to the wiki "Building-the-Emulator", the given command `make linux` fails

i was able to progress the build further with `make all` instead; but that also failed - however, after that failed, i tried `make linux` again, and then it succeeded - the `bin/neo` binary seems to run properly - i suppose that what i have is incomplete and/or partially pre-compiled though, not entirely re-built from source

```
$ make linux
make -B -C kernel clean
make[1]: Entering directory '/code/neo6502-firmware/kernel'
rm -f neo6502.inc  
rm -f build/*
make[1]: Leaving directory '/code/neo6502-firmware/kernel'
make -B -C basic clean
make[1]: Entering directory '/code/neo6502-firmware/basic'
rm -f build/*.*
rm -f sources/generate/*.*
rm -f _include.inc
rm -f test.bas
rm -f memory.dump
make[1]: Leaving directory '/code/neo6502-firmware/basic'
make -B -C emulator clean
make[1]: Entering directory '/code/neo6502-firmware/emulator'
rm -f neo 
rm -f src/*.o 
rm -f src/core/*.o 
rm -f framework/*.o
rm -f 6502/*.*
rm -f /code/neo6502-firmware/build_env/../firmware/common/sources/interface/*.o
make[1]: Leaving directory '/code/neo6502-firmware/emulator'
make -B -C emulator elinux
make[1]: Entering directory '/code/neo6502-firmware/emulator'
echo make -C /code/neo6502-firmware/build_env/../firmware prelim
make -C /code/neo6502-firmware/build_env/../firmware prelim
make -C /code/neo6502-firmware/build_env/../firmware prelim
make[2]: Entering directory '/code/neo6502-firmware/firmware'
make -C ../kernel
make[3]: Entering directory '/code/neo6502-firmware/kernel'
mkdir -p  build
rm -f /code/neo6502-firmware/build_env/../bin/binary.h
rm -f build/binary.h
rm -f /code/neo6502-firmware/build_env/../bin/kernel.bin
python3 scripts/vectors.py /code/neo6502-firmware/build_env/../firmware/common/include/interface/memory.h
64tass -q -b --mr65c02 -Wall kernel.asm -o build/kernel.bin -L build/kernel.lst 
python3 /code/neo6502-firmware/build_env/../basic/scripts/hconvert.py build/kernel.bin build/kernel_binary.h kernel
cp build/kernel.bin /code/neo6502-firmware/build_env/../bin/
cp build/kernel_binary.h /code/neo6502-firmware/build_env/../bin/
cp neo6502.inc /code/neo6502-firmware/build_env/../bin/
cp neo6502.inc /code/neo6502-firmware/build_env/../documents/release
make[3]: Leaving directory '/code/neo6502-firmware/kernel'
cp /code/neo6502-firmware/build_env/../bin/kernel_binary.h /code/neo6502-firmware/build_env/../firmware/common/include/data
cp /code/neo6502-firmware/build_env/../bin/basic_binary.h /code/neo6502-firmware/build_env/../firmware/common/include/data
cp: cannot stat '/code/neo6502-firmware/build_env/../bin/basic_binary.h': No such file or directory
make[2]: *** [Makefile:40: kernel] Error 1
make[2]: Leaving directory '/code/neo6502-firmware/firmware'
make[1]: *** [Makefile:157: elinux] Error 2
make[1]: Leaving directory '/code/neo6502-firmware/emulator'
make: *** [Makefile:79: linux] Error 2
```
```
$ make all
Downloading Raspberry Pi Pico SDK
PICO_SDK_PATH is /code/neo6502-firmware/firmware/build/_deps/pico_sdk-src
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
Defaulting PICO target board to pico since not specified.
Using board configuration from /code/neo6502-firmware/firmware/build/_deps/pico_sdk-src/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.11 (found version "3.11.6") found components: Interpreter 
TinyUSB available at /code/neo6502-firmware/firmware/build/_deps/tinyusb-src/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /code/neo6502-firmware/firmware/build/_deps/pico_sdk-src/lib/btstack
cyw43-driver available at /code/neo6502-firmware/firmware/build/_deps/pico_sdk-src/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /code/neo6502-firmware/firmware/build/_deps/pico_sdk-src/lib/lwip
mbedtls available at /code/neo6502-firmware/firmware/build/_deps/pico_sdk-src/lib/mbedtls
-- Configuring done (23.2s)
-- Generating done (0.1s)
-- Build files have been written to: /code/neo6502-firmware/firmware/build
cmake --build build --target clean
make[3]: Entering directory '/code/neo6502-firmware/firmware/build'
make[3]: Leaving directory '/code/neo6502-firmware/firmware/build'
make -C build
make[3]: Entering directory '/code/neo6502-firmware/firmware/build'
[  0%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  1%] Linking ASM executable bs2_default.elf
arm-none-eabi-gcc: fatal error: cannot read spec file 'nosys.specs': No such file or directory
compilation terminated.
```
just to highlight these lines from the above log:
```
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - skipped
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ - skipped
```
is this correct, that is is looking for and apparently using an ARM cross-compiler?

paulscottrobson, 2024-02-11:
First one is my fault ; I was cleaning up files that are automatically generated and didn't do a clean pull/build. 

It does indeed use an ARM cross-compiler ; it's built on Ubuntu on github.  Have to think about this one :)

astralaster, 2024-02-11:
I've already updated the wiki. It now says run make instead of build_all.sh. You currently need to build everything before building specific targets. I think we need to improve the makefile here.

paulscottrobson, 2024-02-11:
The firmware is currently dependent on Basic and the minimal Kernel, it loads them both at run time.

bill-auger, 2024-02-12:
notwithstanding:
> arm-none-eabi-gcc: fatal error: cannot read spec file 'nosys.specs': No such file or directory
```
$ find -name nosys.specs 

```
what to do about that?

bill-auger, 2024-02-12:
ok a few more notes:

if `make` now always build both emulators, then "Building-the-Emulator" instructions are no longer needed?
> To build the emulators use either make linux or make windows ; clean is not required, this is done for you.

i noticed today that the build requires 'python-gitpython' - im not sure why; because i dont remember needing it last week - maybe that was recently added?

also, im not sure what the "Python Serial library" is either - maybe i have it, maybe i dont - what is it's actual project name or package name?

astralaster, 2024-02-12:
The gitpython package is needed to get the current git tag / commit info into the firmware prompt on start. This was added very recently and needs to be added to the wiki. Maybe the wiki should just point to the shell.nix file or the github workflows, because those contain the dependencies and work.

As you can see the github workflows don't install any python serial library either. So I think its not needed anymore. Maybe @paulscottrobson can confirm this?

bill-auger, 2024-02-12:
> pyserial

ok thanks - i do not have that installed; but the build did not complain about
it - i suppose that one is considered "optional" - or maybe i would need to get
the build to progress further before hitting that requirement

bill-auger, 2024-02-13:
so as a concrete suggestion, i would make these changes:
```diff
-Python Imaging Library / Pillow
-Python Serial library
+Python 'pillow'
+Python 'gitpython'
+Python pyserial (optional, for uploading via serial port)
```

astralaster, 2024-02-13:
> pyserial is required if you wish to upload into the real hardware via the serial port.

You mean uploading files to a running neo6502? Because uploading the firmware is done by openocd.

astralaster, 2024-02-13:
> so as a concrete suggestion, i would make these changes:

I've updated the wiki.

paulscottrobson, 2024-02-15:
Yes, that's exactly it. It is a very simple packet loading protocol, it seems to work fine. Some people like to cross develop directly onto the hardware and swapping the USB key/SD cards would drive me crackers. 

It's not required for anything else.

closed 2024-02-18 by paulscottrobson

bill-auger, 2024-02-28:
i wish you would not have closed this ticket - it was not so much about the documentation as the fact that the software does not compile on my OS - i still get this error:
> arm-none-eabi-gcc: fatal error: cannot read spec file 'nosys.specs': No such file or directory

bill-auger, 2024-02-29:
ok, i got past that error - the wiki is a missing dep - maybe debian systems install it as a dep of the cross-compiler, and that is why it was not noticed sooner; but other systems do not - i needed to install 'arm-none-eabi-newlib'
> [100%] Built target firmware

now i get
> make -B -C cross-compile
> ...
> x86_64-w64-mingw32-g++: Command not found

i suppose this is the `make windows` target? - the wiki does not mention mingw either; but it is reuqired if running the suggested build command `make`

however, i was able to run `make linux` successfully after the failure