paulscottrobson/neo6502-firmware #98: C applications can print only CAPS

bill-auger, 2024-02-11:
upper-case alpha chars passed to puts() are not displayed (or the FG color is the same as the BG color) - lower-case alpha chars are displayed, but as upper-case - however, i discovered that if a buffer is filled with ASCII values as raw ints, the entire character set is accessible

this is demonstrated by the C hello example https://raw.githubusercontent.com/bill-auger/neo6502-firmware/wip-examples/examples/C/hello.c

paulscottrobson, 2024-02-11:
I'll have to look at the produced code and see exactly what it's doing.

paulscottrobson, 2024-02-11:
It's definitely to do with the -t option which is playing games with string constants. The all caps is because the default is C64 is PETSCII I think. I tried it with osic1p which is a very vanilla machine, and none, and it persists in lower casing the 'H'. This is lower case in the binary so it's something to do with the compiler. 
There doesn't seem to be an option for 'leave it alone and do what you're told' !

paulscottrobson, 2024-02-11:
Link is broken btw.

bill-auger, 2024-02-11:
tea, i added the URL preemptively; so that i could reference the ticket in the source code #99

giovannipozzobon, 2024-02-11:
I tried using the option "-t apple2" and the  chars are displayed correctly. I'm not sure that this is the correct answer to the problem.

paulscottrobson, 2024-02-11:
I've asked about this one on the cc65 discussion area. I suspect it's connected with the cl65 defaults - the default type being C64 explains the capitalisation - but why it is lower case is beyond me. It's in the binary as such, so I think it's a compiler problem (or more likely, me not understanding how to invoke it problem)

paulscottrobson, 2024-02-15:
This is now working mostly. (examples/semi-working-cc65). The one thing that isn't is that \n in a string encodes to $0A not $0D. So it does a line feed. 
It still doesn't work split up.

bill-auger, 2024-02-29:
'\n' is the line feed char (it should be ASCII 10) - what it is not doing is carriage return ('\r' ASCII 13); but that technically is correct behavior - it is behaving like windows - *nix systems do the carriage return implicitly along with each '\n' - windows OTOH requires both explicitly - i have it working as expected now, by inserting a '\r' when each '\n' is detected in the string - probably, that is what CC65 should do in its puts() implementation

```C
int write(int /* fildes */ , const unsigned char* buf , unsigned count)
{
  while (count--)
  {
    while(*API_COMMAND_ADDR) {}

    // kludge for BUG: https://github.com/paulscottrobson/neo6502-firmware/issues/98
    if (*buf == 10)
    {
      *API_FUNCTION_ADDR   = API_FN_WRITE_CHAR ;
      *API_PARAMETERS_ADDR = '\r'              ;
      *API_COMMAND_ADDR    = API_GROUP_CONSOLE ;
    }

    *API_FUNCTION_ADDR   = API_FN_WRITE_CHAR ;
    *API_PARAMETERS_ADDR = *buf++            ;
    *API_COMMAND_ADDR    = API_GROUP_CONSOLE ;
  }

  return 0 ;
}
```

there seems to be a regression though - now, the Function 2,7 "set cursor position" does not work as it did before - it obeys the Param[0] X (column); but the Y (row) is always the top row, regardless of the value of Param[1]

closed 2024-03-28 by paulscottrobson