paulscottrobson/neo6502-firmware #110: Building for Mac

omiq, 2024-02-12:
Seeing as it builds on Linux there is no reason it shouldn't be *possible* to build on Mac, so I thought I would try.

Seems to get quite far, errors out on 

```/Users/chrisg/neo6502-firmware/build_env/../firmware/common/include/interface/font_5x7.h:17:56: error: argument to 'section' attribute is not valid for this target: mach-o section specifier requires a segment and section separated by a comma
const unsigned char __attribute__((aligned(4), section(_IMG_ASSET_SECTION ".font_5x7"))) font_5x7[] = {
                                                       ^
```

paulscottrobson, 2024-02-12:
This is to do with aligning font data I think. For the emulator (which is what I presume you're trying to build ?) it could just be a normal constant data array

omiq, 2024-02-12:
Yep, I am hoping to get the emulator running on Mac, and if I can then we could get [TRSE](https://lemonspawn.com/turbo-rascal-syntax-error-expected-but-begin/) compiling for Neo6502 as a target :)

paulscottrobson, 2024-02-12:
That'd be great. If you need any help let me know. I think problems are because we've tended to use #IFDEF LINUX and #IFDEF WINDOWS ; it is basically just SDL2 so it should build easily enough on a Mac.

omiq, 2024-02-12:
Ooh wouldn't it be nice if it is just some missing definitions! :)

giovannipozzobon, 2024-02-12:
I have a Mac too. I'm very interested on. So I could help on this task.

omiq, 2024-02-12:
Woot!

[image]

omiq, 2024-02-12:
Pull Request submitted

https://github.com/paulscottrobson/neo6502-firmware/pull/112

paulscottrobson, 2024-02-15:
Hi omiq. It seems the rebase has completely broken your PR unfortunately so I have closed it. 

Could you resubmit the changes you made - I have already committed the one that was causing the problems, the complex and unnecessary constant definitions.

Thanks, Paul.

omiq, 2024-02-15:
ok will do

paulscottrobson, 2024-02-15:
Now works, I think.

closed 2024-02-15 by paulscottrobson