paulscottrobson/neo6502-firmware #113: NeoUpdateSprite: anchor option have opposite directions in documentation

zbyti, 2024-02-12:
[image]

`NeoUpdateSprite(0, centerX + 3 * 16 + 8, centerY + 3 * 16 + 8, 0, 0, 6);`

looks more like `4` then `6` ;)

paulscottrobson, 2024-02-13:
Are you sure ? If the tilemap (?) is at (0,0), then the draw position should be the centre of tile (3,3) which is the one immediately to the right of the graphic with a lightning sign on it. The anchor position is the position on the sprite used to set its position, e.g. if you move it there the sprite uses its right centre side as its anchor position, so the sprite should move to the left. The documentation could be clearer ....

zbyti, 2024-02-13:
maybe I don't understand the idea... 

```delphi
joy := NeoGetJoy(1);

if joy and 1 <> 0 then anchor := 4;
if joy and 2 <> 0 then anchor := 6; 
if joy and 4 <> 0 then anchor := 8; 
if joy and 8 <> 0 then anchor := 2; 
```

[anchor.zip](https://github.com/paulscottrobson/neo6502-firmware/files/14258701/anchor.zip)

check my binary

paulscottrobson, 2024-02-13:
Yes, it's my fault it needs clarifying. Assuming it behaves the same. When you press left, the anchor point becomes 4. This is the left hand side of the sprite. So when you draw to whatever, it puts the left hand side of the sprite at this point, as opposed to the centre. So it moves to the right.

If you look here https://developer.apple.com/documentation/spritekit/skspritenode/using_the_anchor_point_to_move_a_sprite for example, at the picture of the rocket. In the lower picture the anchor point is at the top. If you put that anchor point in the same draw position as the top one, the whole thing will move down.

PR #116 (2024-02-13): Updated Anchor documentation
The explanation of anchors was a bit rubbish. Fixes #113
closed 2024-02-13 by paulscottrobson