paulscottrobson/neo6502-firmware #115: Red video distortion when access USB filesystem

astralaster, 2024-02-13:
Current firmware produce red visual distortions (lines) when the filesystem is accessed via USB. Its currently unknown why exactly this happens.

zbyti, 2024-02-13:
when I used a USB2.0 pendrive data was read very slowly, it was very visible.

now I use a USB3.0 pendrive and it is less bothersome, but it occurs,  something like using keyboard on the ZX80 :D

paulscottrobson, 2024-02-13:
I suspect this is activity on the part of tinyUSB, maybe locking out the other core ? Needs an RP2040 expert as such is way beyond my knowledge of it.

paulscottrobson, 2024-02-14:
Also suspect #66 is related to this.

astralaster, 2024-03-27:
I did some more testing today. Its now pretty obvious that the problem is flash access. I did remove the PICO_FLASH_SPI_CLKDIV=4 and the boot_stage2 stuff and it does not show the red line glitches at all. We need to reevaluate if its feasible to use PICO default settings for the flash. Otherwise we could try to force specific functions into ram to avoid those flash accesses.

paulscottrobson, 2024-03-28:
Don't want to give up RAM to that unless it's a small amount. It's a bit in short supply as it is.  I suppose the only real way to try it out is to let people try it and see if it works ?

Also found this https://github.com/raspberrypi/pico-sdk/issues/1304 which suggests setting PICO_XOSC_STARTUP_DELAY_MULTIPLIER as a fix to a similar problem ? I think booting problems were the issue ?

OrangeBox72, 2024-07-15:
FYI - (novice) - I assume when you say 'USB Filesystem', you are referring to 'USB' in general and not specifically  USB-Drive/File access.
I wrote a small 'Turtle' program that randomly draws lines at **high-speed** on the screen.
The program does not accept any Input and only draws on the screen.

I can replicate the 'RED line' issue by tapping any key on the keyboard.  The issue will happen approximately 5-10% of the time I tap a key on the keyboard.
I have also (after loading the program from USB-Drive) removed the USB-Drive from the equation by unplugging it.
The issue still manifests with nothing but a keyboard and USB hub attached.

I hope this information is informative.

paulscottrobson, 2024-07-16:
Thanks. It's something to do with tinyUSB undoubtedly, but I'm not quite sure what can be done about it.

astralaster, 2024-07-16:
I don't think it is specific to tinyusb. It's just the cache which is too small to hold all the code and the flash which is too slow to supply it on demand. Tinyusb is just one big chunk of code which needs to run all the time.

paulscottrobson, 2024-07-16:
Figures. There was a bug where BASIC couldn't be loaded from Flash - it would crash. If you read it in chunks and added a delay to let the cache catch up, it worked.

I don't suppose anything can be done about this ?