paulscottrobson/neo6502-firmware #120: SpriteUpdate (6.2) crash emulator

zbyti, 2024-02-14:
```delphi
NeoUpdateSprite(0, posX, posY, 0, crash, 0);
```

if `crash` indicates non existing sprite (for example I exported 1 and chose 2) then emulator crash.

paulscottrobson, 2024-02-15:
Can you supply the context for this (e.g. files ?). I've tried it with illegal sprite numbers and it does what I expect, e.g. in BASIC it gives an out of range error. I need to know what the actual parameters are, and what the graphics file is. Running the below code ; $80+i also gives that error (there are only 4 sprites in the file).

```
cls:sprite clear:gload "graphics.gfx"
for i = 0 to 5
	sprite i image i to i*40+20,50
next
```

zbyti, 2024-02-15:
thanks for checking. I can't reproduce this bug on v0.12.0, problem solved by update ;)

closed 2024-02-15 by zbyti

paulscottrobson, 2024-02-15:
Okay. Slightly weird, I think there was something broken in 0.11.0 because heard of another couple of problems that magically vanished.