paulscottrobson/neo6502-firmware #152: Get USB Joystick/Gamepad working

paulscottrobson, 2024-02-18:


astralaster, 2024-02-21:
Do you already have an idea on how the API should look like? I think it would be great to not limit the gamepads on a specific button count or similiar.

paulscottrobson, 2024-02-23:
Early experiments umpromising. Have two joysticks (a faux XBox controller, and a classic joystick), both of which work, neither are detected (e.g. mount does not appear to be called). More work required on this.

paulscottrobson, 2024-02-23:
> Do you already have an idea on how the API should look like? I think it would be great to not limit the gamepads on a specific button count or similiar.

Well, there can be up to 8 fire buttons supported as is ; the joypad() return value is a bit pattern, so it can be masked or simply used as "any button" by testing for non-zero.

astralaster, 2024-02-23:
I think to honor the neo in neo6502, it should support a standard xbox like controller. So analog input and all buttons.