paulscottrobson/neo6502-firmware #160: buggy makeimg.zip

zbyti, 2024-02-20:
[image]
[image]
[image]

`python makeimg.zip`

gives:

```
Image Set Creator
	Tiles   16x16 found 41
	Sprites 16x16 found 6
	Sprites 32x32 found 0

```
I expect 48 tiles not 41.

paulscottrobson, 2024-02-20:
It's difficult to see. The way makeimg works is it does not include blank ones. Some of those look blank. I presume you want them blank for game reasons ? If it doesn't matter then just put something in them and your numbers will line up, you'll just waste a bit of memory. If they must be blank for game reasons, send me your files and I'll have a think about a way I can work round it.

zbyti, 2024-02-20:
And how can I waste memory that I don't have direct access to? If I don't use 128 tiles, I won't be able to do anything with that memory anyway, right? ;)

zbyti, 2024-02-20:
```python
	def isBlank(self):
		return len([x for x in self.data if x != 0x88]) == 0
```

this is correct in may case?

zbyti, 2024-02-20:
```python
class Tile16x16(object):
	def __init__(self,data):
		self.data = data
	def getSize(self):
		return 16
	def getData(self):
		return self.data
	def isSprite(self):
		return False
	def isBlank(self):
		return False
```
this change works as I expected, I don't need placeholders... I don't care about empty placeholders because I can't reuse this GFX RAM and I have numbers in the tilemap corresponding to the tilesheet :)

paulscottrobson, 2024-02-20:
The only downside is that memory is shared between sprites and tiles. so if you have lots of them you eventually will run short.

closed 2024-02-20 by paulscottrobson