paulscottrobson/neo6502-firmware #163: Fix colors of neo6502 logo

astralaster, 2024-02-21:
The colors of the neo6502 logo on bootup are meant to be different then they look right now. It seems to be limited by the console palette right now.

[image]

paulscottrobson, 2024-02-21:
Okay .... I'll try to figure it out .... this is harassment :(

paulscottrobson, 2024-02-21:
So what colours do you want what .... give me some RGB values and I'll make it work.

That's what I liked about coding for the RCA Studio 2. No colour arguments ...

astralaster, 2024-02-22:
> So what colours do you want what .... give me some RGB values and I'll make it work.
> 
> That's what I liked about coding for the RCA Studio 2. No colour arguments ...

The colors in the image above are the wanted ones. #ff00ff for the spiral and #00ffff for the text.

paulscottrobson, 2024-02-22:
No problem, I think.

closed 2024-02-22 by paulscottrobson