paulscottrobson/neo6502-firmware #175: Implement additional BASIC commands/statements.

leventp, 2024-02-22:
It would be useful to have the following commands/statements for modular programming and convenience:

CHAIN <filename> : Load and run the file specified in <filename>
INCLUDE <filename> : Load the library procedures defined in the file so they can be used in multiple programs
WAIT/PAUSE <period> : Pause execution of a program for the specified period (fraction of a second based on implementation)

paulscottrobson, 2024-02-22:
The first and last are fairly straightforward. The library has both technical issues and issues related to being limited to a 6502 ; limited memory, not easy just to shift stuff about.

leventp, 2024-02-22:
Understood. WAIT/PAUSE is most useful as people use loops and timer to simulate it.

paulscottrobson, 2024-02-22:
I think I'd just extend RUN to allow RUN "<file>" and that would be a composite of LOAD and RUN. One has to be careful adding tokens too enthusiastically ; any token (e.g. WAIT) means you can't have a variable called WAIT any more. Though programs that already exist should work, and be listable, though you couldn't edit them.

I've moved libraries as that's a more complex issue.

leventp, 2024-02-22:
If RUN "" can be embedded into programs, it would be fine. The idea is to split larger programs into smaller ones to save memory.

closed 2024-02-23 by paulscottrobson