paulscottrobson/neo6502-firmware #181: Documentation for the new fileio and tty calls

davidgiven, 2024-02-22:
This is my job, as I wrote this stuff; how would you like it done? I know there are plans to autogenerate the documentation from markup in the dispatch file. Do you want me to do that or edit the ODT now?

paulscottrobson, 2024-02-23:
It would be sensible to do it in the markup but I don't know where that is at, if anything, Do you want to experiment with PR#101 ?

paulscottrobson, 2024-02-28:
Probably a good idea to hold off until this PR (#101) is sorted. The current documentation system is difficult to edit other than one at a time.

paulscottrobson, 2024-02-29:
You may want to review this ? Looks like Bill A has done a lot of the work.

paulscottrobson, 2024-02-29:
This file https://github.com/paulscottrobson/neo6502-firmware/blob/main/firmware/common/config/system/group3_fileio.inc

bill-auger, 2024-02-29:
yes, i add quite a lot of detail - i was not only reforming the workflow, but trying to understand the system in the process - i encourage those who know the inner-workings better than i, to read the entire api.pdf - i did a fair amount of interpreting the source code and guessing - especially note the table rows with 'Q:' comments - im sure that a got a few things wrong

davidgiven, 2024-03-01:
Right, I'll take a look. I'm happy to do all the docs for the console and file stuff I've added.

I see that the parameters are currently described in the text body, rather than as a table or list. Is that the preferred house style or has this just not been thought about yet?

Also, markup in the documentation section is Latex, right?

paulscottrobson, 2024-03-01:
Yes, though you don't need to know very much ; most of the work is done by bill auger, there are macros for parameter display, you pretty much just type in what each function does ; the outline is all there. The parameters in the text body looks like its the way to go :)

paulscottrobson, 2024-03-01:
If you are a busy man I can write it and you can review it. I'm retired so I have the time to do this mostly :) 

I did add comments for your Console Changes, largely to the effect of 'don't touch these'. There are some firmware routines that are just support routines and shouldn't really be used.

bill-auger, 2024-03-01:
the function tables are compiled from the GROUP->FUNCTION->DESCRIPTION in dispatch.config - those should be brief (less then 5 lines) to keep the size of the tables reasonable - just mention the basics of the parameters and outputs, something of a quick-reference cheat-sheet - if any need elaboration, most of the function groups have a dedicated section later in the document - those elaborate descriptions are in the api.tex.in template

the descriptions of the console and file i/o functions were not very verbose, so i did not give them a dedicated section; but that is easy to do, for example:

```
\pagebreak
\section{File I/O}\label{file-io}

Plain text here...
```

closed 2024-03-01 by paulscottrobson