paulscottrobson/neo6502-firmware #210: No CopyFile routine in fileInterface

bocianu, 2024-02-24:
I am not able to copy the file using the current API. 
There is a delete and rename function, but no copy capability.

paulscottrobson, 2024-02-25:
Could be done using open/read/write/close obviously but probably a good addition as more than one use. @davidgiven is this a useful add on

davidgiven, 2024-02-25:
The question is, do you want progress information or not? If so, it has to go through the 6502, and if not, it can happen entirely on the RP2040 side. The filesystem is fast compared to the 6502 but not _that_ fast, at least with the USB stick I'm using, so copying big files could take a second or so.

Disadvantages of the former is that you'd have to allocate a buffer somewhere in 6502 memory. Disadvantages of the latter is that you'd have to allocate a buffer somewhere in RP2040 memory! There is some spare space on the heap but I'd have to check how much.

I think having a function on the RP2040 side does make sense (as it would allow a `*COPY` which doesn't overwrite the program; Acorn, I'm looking at you here), but for the specific purpose of a file manager, you might want to do it yourself to get a better user experience.

paulscottrobson, 2024-02-25:
I wouldn't think so ; files aren't going to be big enough to make a progress bar worth while, even if wifi comes online.

PR #219 (2024-02-27): Add an API call for copying files.
It's even tested!

Performance on my rather bad USB stick is ~4-6 kB/s, which is miserable, but probably fine for this platform. It's using a 1kB buffer allocated when needed from the heap (and then freed again). We could probably make this bigger but let's leave it like this for now until we know how much spare heap we're going to have.

Fixes: #210
closed 2024-02-27 by paulscottrobson