paulscottrobson/neo6502-firmware #211: I'm unable to use relative paths.

bocianu, 2024-02-24:
While coding my Zion Commander I noticed that in order to load files I always have to use the absolute path.
But this also applies to all programs that try to run from a directory other than the root directory. 
If they use other resources that are loaded after running them, they must either be in the root directory, 
or they must be pointed to using an absolute path. 

Without working relative paths, grouping programs in folders with their resources is severely hampered. 

I think a simple solution could be to add to the API the ability to define a CWD - current working directory.
And then add its support to functions that operate on files.

paulscottrobson, 2024-02-25:
@davidgiven one for you ?

bocianu, 2024-02-25:
> There is CWD tracking, but you have to explicitly use function 3,15 (ChangeDirectory) to change it. Also, I don't think anyone's tried it yet. I think it does make sense to change directories in the loader to the one containing the binary, but that's strictly a loader issue, so right back at you, @paulscottrobson !

o thanks! I will try using 3,15 and let you know.

bocianu, 2024-02-26:
Looks like it's working! Thanks for guiding me on the right path.
The issue can be closed.

closed 2024-02-26 by bocianu