paulscottrobson/neo6502-firmware #289: Documentation: Messaging API for Mathematics Functions does not document Register1/Register2

asiekierka, 2024-03-10:
To clarify: it does not say which Params they are stored in, or what format they are.

paulscottrobson, 2024-03-11:
They are really BASIC helpers, which is the primary reason they aren't documented. It's a slightly odd format, because it's designed to be used in the internal BASIC stack and possibly accessible by others. What do you want to use it for ?

asiekierka, 2024-03-11:
It stuck out to me while I was going through the API documentation; there is no specific reason for me seeking out the math functions, but I was evaluating which functions could be exposed in what ways as part of [the LLVM-MOS target](https://github.com/llvm-mos/llvm-mos-sdk/pull/323).

paulscottrobson, 2024-03-11:
They're really BASIC helpers.

I don't know enough about it. It does mathematics for 32 bit integers and C format floats (the standard 4 byte format). Does LLVM MOS mostly use 16 bit integers (in practice) ? 32 would likely be very slow.

If this is the case my initial inclination would be to provide specific functions for 16 bit multiply and possibly divide.  Could do 32 bit if they are used. This would be cleaner and could also be used for other languages

As for the other functions, add and subtract integers are actually quicker in 65C02 code (the call takes longer) and floats probably have fairly minimal use ?

asiekierka, 2024-03-11:
> Does LLVM MOS mostly use 16 bit integers (in practice) ? 32 would likely be very slow.

Yes, but in theory nothing stops us from sign-extending 16-bit integers into 32-bit integers. I suppose there could be a value somewhere to control operand size/type.

In general, LLVM-MOS provides everything that the C standard does, including large integers and software floats (albeit slow). It is possible, somewhat easily, to overwrite the library functions LLVM's compiler runtime provides (for multiply, divide, floating-point operations, other math functions...) with ones that accelerate these operations using the messaging API.

They have minimal use, sure (unless a program decides to use them!), but what's the fun if not trying to support *all* of a given hardware stack's functionality?

PR #298 (2024-03-13): Added note on Mathematical data
A brief explanation of how developers can access the mathematical helpers in the firmware.

Fixes #289
closed 2024-03-13 by paulscottrobson