paulscottrobson/neo6502-firmware #385: chars - right bottom corner ruin my code ;)

zbyti, 2024-03-30:
https://github.com/tebe6502/Mad-Pascal/blob/master/samples/neo6502/demoeffects/plasma.pas

I wrote workaround but one char left behind ;) no one likes left someone behind :D 

binary attached: [plasma_0x1000.zip](https://github.com/paulscottrobson/neo6502-firmware/files/14813105/plasma_0x1000.zip)

zbyti, 2024-03-30:
[bad_0x1000.zip](https://github.com/paulscottrobson/neo6502-firmware/files/14813115/bad_0x1000.zip)
if i try full screen :]

zbyti, 2024-03-30:
I changed `write(row)` to `NeoDrawString(0, y * 7, row);` and now is faster and without last character bug.

paulscottrobson, 2024-03-31:
What's the exec address of this binary ?

paulscottrobson, 2024-03-31:
Also the program is the drawString version ?

zbyti, 2024-03-31:
`write` version with workaround (last row have one char less the then others) : https://github.com/tebe6502/Mad-Pascal/blob/a58dba631941b7e035a6820e01a52c43baed7a35/samples/neo6502/demoeffects/plasma.pas

`DrawString` no bugs, fastest: https://github.com/tebe6502/Mad-Pascal/blob/master/samples/neo6502/demoeffects/plasma.pas

zbyti, 2024-03-31:
> What's the exec address of this binary ?

$1000, 0x1000 ;)

`neo plasma_0x1000.bin@1000 run@1000`

paulscottrobson, 2024-04-01:
The problem is the console system is designed to operate like one, so it is a print character and advance system. So if you print on the last character of a line it will move to the next character position, the first line on the next line down, and if you are on the bottom of the line this involves scrolling. It could be changed to delay the move till printing but that may cause other issues (e.g. the cursor is not where you'd expect it to be after printing a character). 

For this specific usecase I'd suggest making it 53x29 and using the bottom line for a text comment.

erlendnagel, 2024-04-01:
I guess using text 'x' to 313,231 wouldn't trigger the scrolling.