paulscottrobson/neo6502-firmware #399: "clrscr" API 2.12 reset value of "SetSolidFlag"

zbyti, 2024-04-02:
I assumed that if I `SetSolidFlag(1)` on the beginning of my program I always draw solid rectangle but after `clrscr` I need set this flag again.

paulscottrobson, 2024-04-02:
Clearing the screen should initialise everything to a default state

closed 2024-04-02 by paulscottrobson

zbyti, 2024-04-02:
I don't agree because we have:

```
procedure NeoSetDefaults(acol,xcol,solid,size,flip:byte);

Sets colour for upcoming drawing operations.
parameters:
acol (byte) - value that will be ANDED with current colour
xcol (byte) - value that will be XORED with current colour
solid (byte) - solid flag (0=empty, 1=solid)
size (byte) - dimension (for text)
flip (byte) - flip flag for tiles (0=none, 1=horizontal, 2=vertical, 3=both)
```
clear screen should clean only screen, but you're the Boss ;)

paulscottrobson, 2024-04-04:
Well, perhaps, yes. You could make a case for it being just that or a generic reset of the graphics system. Problem with it is is I don't really want to make potentially breaking changes unless it's necessary. There's several things I would probably change if it was "just me".