paulscottrobson/neo6502-firmware #481: SD card problem started in release 0.15.2

RocketManRC, 2024-05-01:
I noticed the discussion on Discord about the SD card not working properly any more and I have discovered that the problem started with release 0.15.2. (i.e. with 0.15.1 the SD card works fine but not with 0.15.2 or later). I'm using the Olimex card that plugs into the EXT connector.

In the version with the fault the cat command will show the files but after loading a .bas file cat will no longer work and running the program will give an I/O error where it tries to load the .gfx file.

I looked at the change log for 0.15.2 and nothing looks to me like it would affect the SD card access but then again I'm just starting to get familiar with the way the firmware works.

This is a great project with a lot of amazing development work to get it where it is!

paulscottrobson, 2024-05-01:
Thanks, I will have a look.

paulscottrobson, 2024-05-02:
The only thing I can see that would affect the SD Card operation is the change to clear.asm https://github.com/paulscottrobson/neo6502-firmware/compare/v0.15.1...v0.15.2 ; this does sort of make sense. 

CLEAR, which is the general BASIC reset, resets the I/O system. I'm wondering if this breaks SD Card access. I probably tested it at some point but maybe not multiple times the same run.

I would suggest the first thing to do is to remove these added lines (97-102, calling the I/O Reset). Attached is a release without that code. Unfortunately my SDCard adaptor is still missing.
[neo6502.zip](https://github.com/paulscottrobson/neo6502-firmware/files/15184676/neo6502.zip)

 If this fixes it then I will make a better fix which effectively disables those pins on the SDCard version. This is no actual loss, because you can't use them anyway :)

[Uploading neo6502.zip…]()

RocketManRC, 2024-05-02:
That definitely fixes the problem as it works perfectly now. One thing that's different is that the screen goes black while a file is loading. Is that the expected behaviour?

paulscottrobson, 2024-05-02:
If this doesn't work let me know ! Still can't find the ruddy SD Card.

closed 2024-05-02 by paulscottrobson