paulscottrobson/neo6502-firmware #489: Switch from Wozmon to EWoz

rprouse, 2024-05-12:
This is just a suggestion in case it interests you. EWoz is a slightly improved version of WozMon. I have a version of it running as the OS (along with BASIC) for a 6502 computer I built and the enhancements are useful. I found the code in this Gist, https://gist.github.com/BigEd/2760560 and quite a few people over at 6502.org use it for their computers.

From the Gist, the enhancements are,

- It's using ACIA @ 19200 Baud. (will need to be changed)
- It prints a small welcome message when started. 
- All key strokes are converted to uppercase. 
- The backspace works so the _ is no longer needed. 
- When you run a program, it's called with an JSR so if the program ends with an RTS, you will be taken back to the monitor. 
- You can load Intel HEX format files and it keeps track of the checksum. 
- To load an Intel Hex file, just type L and hit return. 
- Now just send a Text file that is in the Intel HEX Format just as you would send a text file for the Woz mon. 
- You can abort the transfer by hitting ESC. 

I saw in Discord that you were thinking of replacing Wozmon with an enhanced monitor. I agree with the sentiment, but I thought this might be a stepping stone. Personally, I'd prefer to see a more full featured monitor, assembler, disassembler, but that could take time and this might fill some of the gap in the meantime.

My thinking is that the serial upload could be modified to allow uploading Intel HEX files for a quick cross-platform assembly workflow.

If this is not of interest, please close the issue. No hard feelings 😸 

It would take more work because it is C64 centric, but there is also Butterfield's SuperMon+64, https://github.com/jblang/supermon64

paulscottrobson, 2024-05-14:
Thanks, I will have a look. There is a slight issue with the serial in that it doesn't seem to work in a sort of sleep mode. The original design responded to the serial port automatically, and I couldn't make that work reliably.

superian, 2024-07-31:
Those of us who had Atari 800s will remember DDT ("Dunion's Debugging Tool") which, thanks to the Atari hardware, had a separate screen for its output - you could switch between your program's graphical output and the text of the debugger with a single keystroke. As with the Mac/65 assembler, it was expanded to add the 65C02 instructions.

[DDT on AtariWiki](https://atariwiki.org/wiki/Wiki.jsp?page=DDT)

Given how the display is generated on the Neo6502, having something similar would be extremely useful.

paulscottrobson, 2024-08-03:
The DDT thing is interesting. I'll give these ideas some consideration. One issue is the emulator has a debugger built in, which does provide this alternating text/graphics already. It's a very accurate emulator (the timing is slightly out, but the behaviour is very consistent because it's mostly running the same code).

closed 2024-08-03 by paulscottrobson