paulscottrobson/neo6502-firmware #513: Blitter broken since 22 May

bocianu, 2024-06-15:
Merge of new blitter from @bcampbell has broken (limited) the API.
I cannot reproduce same functionality I had on previous one on this new one.
My two programs stopped working.

paulscottrobson, 2024-06-15:
That may be my fault. I did give @bcampbell a fairly free hand in that as when he changed it it was very new beyond the very limited simple copy. Is it basically not working *at all* or is it something specific ?

bcampbell, 2024-06-15:
> Merge of the new blitter from @bcampbell broke the API. I cannot reproduce the same functionality I had on the previous one on this new one. My two programs have stopped working.

Do you mean "Simple Blit Copy" (Function 2) or "Complex Blit Copy" (Function 3)?

The Simple Blit should have stayed the same, but the Complex Blit did change. I figured it had only been out for a week or so, so there probably wasn't anything relying on it (famous last words :-).
There was a bunch of stuff the API didn't support that I thought it would benefit from, so I went ahead and changed it.

What are you doing with it? The new complex blit _should_ be able to do everything the old one could, even if it's a bit of an arse updating the calls...

bocianu, 2024-06-15:
Yes, I was using function 3 :D 
Short description what was I doing I what I cannot achieve right now:

Previous implementation was able to blit rectangles using this structure:
[image]

And at start of my program I was painting "bricks" on screen and copying them to GFX memory using blitter and this settings below:

[image]

So in gfx memory every brisk was a linear block.
And then in game using the same settings I was blitting it back to screen.
And it worked flawlessly. 

And then the merge came, and I tried to use a new improved blitter to do the same, so new block structure was defined like:

[image]

and tried to achieve the same thing:

[image]

And it blits only the first line of the image. Maybe the problem lies here:

[image]

if width is not used in target, I cannot define different shape as an output. If source is linear I cannot make destination narrower rectangle?

bcampbell, 2024-06-15:
OK, I think something like this should work:
```
blockGFX.format := 0
blockGFX.stride := BRICK_WIDTH;
blockGFX.width := BRICK_WIDTH;
blockGFX.height := BRICK_HEIGHT;

blockScr.format := 0
blockScr.stride := 320;
```

Stride is the number of bytes between the start of each line.
So for your linear source brick, the stride is the same as the byte width of your image.
For the screen, it's 320 bytes between the start of each line. 
The stride allows all kinds of tricks:
You could use a negative source stride to draw your tile flipped vertically.
Or you could use a zero source stride to draw the same source line repeatedly.

BTW, if your source brick image is only 16 colours, you could pack 2 pixels into each byte and the blit operation will unpack it for you on the fly. The changed lines would be:
```
blockGFX.format := 1   // 2 pixels per byte
blockGFX.stride := BRICK_WIDTH/2;
```
Or if you've got monochrome source images you can pack 8 pixels/byte:
```
blockGFX.format := 2   // mono (8 pixels per byte)
blockGFX.stride := BRICK_WIDTH/8;
```

When you've got a total 64KB of RAM to play with, I find 8 bit-per-pixel graphics suddenly seem to take up an awful lot of space :-)
Hope this helps!

bocianu, 2024-06-15:
ok, so stride was not an 1:1 equivalent of the stepoffset 😄 
Now, after the changes you suggested it works! thanks.
And the possibilities are much wider now. 
We can close the issue now.

closed 2024-06-15 by bocianu