paulscottrobson/neo6502-firmware #562: Feature Request: Addition "DOS" commands!

jewettg, 2024-08-04:
Looking for some "DOS" commands that I could use to navigate the USB storage that I have attached.  I found some I/O commands for interacting with the hardware, and the use of "cat", "load" and "save", but one glaring omission was a way to navigate directories?!    I also found this reference sheet: https://neo6502.com/reference/basic/

Is it possible to get some of the basic ones and their functionality?  **I was thinking the minimum of:**

`cat` -- list the contents of the current directory (have √)  (maybe we can consider renaming to ls or ldir, see below.)
`load ""` -- load a file into memory (have √)
`save ""` -- save a file to disk (have √)
`cd` or `prefix`  --- change directory (need)

**Would be nice to have (but it can also be done ):**

`mkdir` -- make a folder
`cat` -- list a file to the console (may need to put some guardrails on this one, so that it does not attempt to list binary that could cause unknown issues; conflicts with "cat" command that lists a directory)
`mv`  -- move a file from folder to folder, or rename a file (could be rename)
`cp`  -- copy a file from to new file or location.
`del` or `rm` -- to delete a file (again, maybe with guardrails, or let people make mistakes, or do not allow wildcards.)

---
In the list of commands above, they would need to support the concept of ".", "..", and "/".
..and to put icing on the top, support wildcards, but with scoping: *=greedy, +=1ormore, and/or some other character to represent a single "any" character.

jewettg, 2024-08-04:
Someone of discord mentioned: https://neo6502.com/wiki/MOS-Interface/
However, I am unfamiliar with this MOS interface?

paulscottrobson, 2024-08-05:
Well the idea was that you could use the mos command e.g. mos "copy "+file1$+" "+file2$ for example. The other alternative is you can type them at the command line, e.g. *cat 

With a mos.copy the parameters would have to be in quotes anyway because of the tokeniser. so it would be more like mos.copy "file1.txt","file2.txt"

The advantage of doing it this way rather than having individual commands (cat is a hangover from very early versions) is the mos commands can be extended without having to add keywords. It is also good for programming in C / Pascal / whatever which works the same way.

Documentation definitely needs improving though.

closed 2024-08-05 by paulscottrobson