paulscottrobson/neo6502-firmware #606: scaling of SOUND frequency is way off, and durations inconsistent

phiber1, 2024-09-10:
I'd be happy to look into this to resolve the issue. But at present, the scaling for the 16-bit SOUND function's frequency in BASIC is pretty bad. The output tone remains the same frequency for several sequential values. And even just beyond a thousand Hz or two, the frequency parameter is way off from what's expected. The frequency is just plain wrong compared to what's output.

Another issue observed, is duration. The duration parameter should be uniform, independent of frequency. While playing an isolated tone, this is generally true, but try for-looping through a step of tones. Even with WAIT, the duration of the component tones is all over the place. There is no consistency despite choosing a fixed duration for each tone in the loop.

paulscottrobson, 2024-09-13:
So, it's partly deliberate. It can be improved somewhat, what's happening is it's generating a square wave using sample frequency / required frequency. This is , I agree, pretty basic, and it's designed with beeps and boops in mind not music. It should however be more accurate than this without ramping up the sample rate.

The sound system is a trade off between CPU speed and Sound generation. What I would like to do is to have a proper FM synthesis emulation of some sort from somewhere and the system is now designed to support this, so one could have a SID emulation in there say, but the CPU speed would be affected. (The simple 4 tone / white noise generator drops the CPU speed from 6.25Mhz to about 6.1Mhz), but the 'sound chip' could be switched on the fly using a BIOS call.

I'm not sure about the duration, this shouldn't be a problem as it's handled by a 20Hz tick routine. The only reason I can think of where it would slow down is if the RP2040 is being really flogged hard with tile maps (especially) and so on, because it does stop effectively to draw these. I will have a look.

I guess you just have a simple FOR loop playing a short tone ?

paulscottrobson, 2024-09-13:
If you want to have a look or have more idea than me feel free to have a look. Sound is not my thing :)

phiber1, 2024-09-13:
I'm all for maintaining the original square wave interface, though I'd probably consider adding a simple waveform selector for triangle/sawtooth/sine presets, but that's just me. Having a SID or Yamaha emulation would definitely be a cut above, and I'd probably include the classic TI 76489 in that list. You'd likely get a lot of interest from the VGM sequencing crowd with that kind of selection.
I'd be more than happy to have a look at the current frequency and duration issue as it stands.
The duration issue is easily reproduced:
for i=1 to 9000:sound 0 clear i,5:next
You can clearly hear the tone durations get much longer as you go up in frequency.
Interestingly, I'm only hearing this in a for loop, and no issue with duration for individual sound commands.

paulscottrobson, 2024-09-16:
I agree entirely about the new waveform system. The current one is designed to be extensible with multiple waveforms, there are currently two square wave and white noise.

As for the duration, I think it's because of the other problem. Sound is in a queue, and sound 0 clear clears that queue before playing the sound, so I think as the frequency increases the divisor issue means more notes are the same.

The noise/square wave code is here https://github.com/paulscottrobson/neo6502-firmware/blob/main/firmware/common/sources/interface/sndcreator.cpp - the obvious fix is rather than divide the sample rate by the frequency and count, scale it up and subtract on wrap around. Would mean the square wave mark/space would vary slightly, but the mean sound should be better ?

PR #616 (2024-09-19): Back to original sound
Single value ; the self adjusting version didn't work with PCM. May need an actual waveform in there.

Fixes #606 

Better sound will be implemented for 1.1 as an alternate ; this is a game beeper.
closed 2024-09-19 by paulscottrobson