paulscottrobson/neo6502-firmware #623: Feature Request: Map exit key by configuration or command line options

linuxplayground, 2025-04-19:
Some applications use the Escape key as part of the application.  The most obvious one I can think of is the QE Text Editor in CPM65.

In the emulator, the ESCAPE key is hard-coded to exit the application unless the `keys` command-line option is given when starting the emulator.

Unfortunately, that option also disables all the debugger keys.  In that instance the CTRL+ESCAPE Key sequence is required to exit the emulator.  But on windows machines, that just results in the default behavior of bringing up the Windows Start menu.    

It would be preferable, I think, to be able to choose which key quits the emulator _or_ have the option to disable quit-by-keypress altogether.  The goal would be to support developers who want to use the ESCAPE key in their games and applications _and_ debug them while developing them.

A super hacky work-around that's working for me is to just select a different key on line [92 of gfx.ccp](https://github.com/paulscottrobson/neo6502-firmware/blob/a8c111e13b67f8bd62007ef5f2a8bff3e1cc95bb/emulator/src/framework/gfx.cpp#L92).  In my case, I mapped it to F3.  That way I can have debugging, a functioning ESCAPE Key and quit-by-keypress at the same time.

Rather than hard-coding all of this, can it be brought out to a config file with sane defaults.?

The MAME approach is quite cool.  You can run the mame emulator and it builds a defaults ini file on first run.  That ini file can be edited and passed in as an option on startup.  I can imagine options for debugger keys, quit, joypad to keyboard mapping and anything else folks want to customize.

Thanks

paulscottrobson, 2025-04-19:
There is a 'keys' command line option which disables these (I think you press Ctrl+ESC instead)

linuxplayground, 2025-04-19:
I know there is. But the keys command line option also disables the debugger, which is not useful if you want to debug software that uses the escape key. 

Also on windows, Control + Escape brings up the start menu.